AcheronProject/AcheronSetup #2: Addressable LEDs

rgbpcb, 2021-12-20:
Should I connect pin 28 (PB15/SPI2_MOSI / "RGB_3V3") of the MCU directly to the DATA_INPUT pin of the first addressable RGB LED (SK6812 mini-E)?

Do I have to transform the voltage? How?

Maybe you could add a few addressable RGB LEDs as an example to the Joker PCB, so that I could move them, add more of them, and so on.

Thank you for this wonderful work!

Gondolindrim, 2022-01-19:
Hi @rgbpcb !

Yes, you cannot connect pin 28 directly to a data input pin of the adressable LEDs because those work with a 5V logic whereas the MCU works with 3.3V. What you need essentially is a level translator. I recommend using TXS0101 . I can add this information to the Joker template.

Gondolindrim, 2022-01-19:
Discussion continued in issue https://github.com/AcheronProject/AcheronSetup/issues/3 .

closed 2022-01-19 by Gondolindrim