AcheronProject/AcheronSetup #3: USB connector

rgbpcb, 2021-12-20:
How do I add a USB connector?

Do I understand correctly that the only thing that's missing in the Joker PCB in order to work are the "USB Connector" and "USB ESD" boxes that I can copy from the Athena schematic?

Maybe you could add a USB connector to the Joker PCB as an example, so that I can move it (and re-route the differential pair carefully).

Note that I'm looking directly at `joker48.kicad_pro`. I couldn't follow the instructions in [the readme](https://github.com/AcheronProject/AcheronSetup/blob/main/keyboard_creator/README.md) because I can't find `core-files`, and `make create` says `make: *** No rule to make target 'create'.  Stop.`, and `./keyboard_setup.sh` says `Initializing git repo... error: refname refs/heads/master not found   /
fatal: Branch rename failed`

Thank you!

Gondolindrim, 2022-01-19:
Hi @rgbpcb !

It was just pointed out that it would be a good idea to indicate on the template how to connect the MCU to a USB and the LDO that steps 3.3V down from 5V. I will be adding those to the [joker article on the Acheron documentation](https://acheronproject.com/joker_mcus/joker/)

rgbpcb, 2022-01-19:
Thank you, I'm looking forward to this!

It would be great if you add the USB and LDO to the PCB files in this repository. You can place the components at any location you want, and I would move them to another location if necessary.

(Same for the Addressable RGB LEDs from issue #2.)

That would be very helpful for seeing what should be connected to what correctly. Thanks!

Gondolindrim, 2022-01-19:
The problem with this whole discussion, from a documentation standpoint, is that the compartmentalization of information is made difficult if I try to add a "features" section. I will see what I can do; in the meantime, please look at the [Athena PCB](https://github.com/AcheronProject/Athena) for design reference.

closed 2022-01-19 by Gondolindrim

rgbpcb, 2022-01-24:
> The problem with this whole discussion, from a documentation standpoint, is that the compartmentalization of information is made difficult if I try to add a "features" section. I will see what I can do

In the documentation, you could write something like _"The Joker PCB files contain a USB connector for your convenience. Feel free to move it to a different location on the PCB, but make sure to route the differential pair carefully, as described [here](http://acheronproject.com/acheron_setup/acheron_setup/#33-overall-design-recommendations)."_

Does that seem good, or what information is problematic?

Gondolindrim, 2022-01-24:
Well, the Joker implementation itself is not particularly problematic; rather, the integration of keyboard features into Joker's documentation.

However I do feel that these features are important. I will try adding them later.