AcheronProject/AcheronSetup #8: Symbol STM32F411-48 used in the joker48_template project but not found in acheron_Symbol

Razerban, 2022-01-26:
The symbol `STM32F411-48` is used in the joker48_template project but is not found in `acheron_Symbols` symbol library.

It's still defined in the schematic file:
https://github.com/AcheronProject/AcheronSetup/blob/38a441b03cf2d87f9b84fae42a58691b0dbfa74a/keyboard_creator/joker48_template/joker48.kicad_sch#L425-L895
 
But when I try to update the symbols from the library, I get the following error:
[image]

I think that the template project has been created with an old revision of the `acheron_Symbols` library which had the symbol `STM32F411-48` in it and KiCad copied the symbol definition when it was placed to the schematic file.

Gondolindrim, 2022-01-31:
Hi @Razerban 

The template does not feature any libraries, symbols or footprints, because the script will add those later.

Razerban, 2022-02-01:
> Hi @Razerban
> 
> The template does not feature any libraries, symbols or footprints, because the script will add those later.

Hey @Gondolindrim!

That's true. The project is after all a template and the libraries would be added by the script.

However, even when using the script to download the libraries (symbols and footprints), the problem persists because the symbol `STM32F411-48` isn't present in the downloaded symbols library `acheron_Symbols`.

[image]

Gondolindrim, 2022-02-01:
Hi @Razerban , 

If you take a close look, the footprint used is `STM_UFQFPN-48_LQFP-48-1EP_7x7mm_P0.5mm_HandSoldering_ThermalReliefs`

[image]

closed 2022-02-01 by Gondolindrim